gokrazy/gokrazy #292: Cannot build with tailscale

andig, 2025-01-25:
Added tailscale per docs:

```json
{
    "Hostname": "evcc",
    "Update": {
        "HTTPPort": "8080",
        "HTTPPassword": "HjJv6aBX51G4wT8EueCc"
    },
    "Packages": [
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/mkfs",
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/wifi",
        "github.com/evcc-io/evcc",
        "tailscale.com/cmd/tailscaled",
        "tailscale.com/cmd/tailscale"
    ],
    "PackageConfig": {
        "github.com/evcc-io/evcc": {
            "GoBuildFlags": [
                "-tags=release -trimpath -ldflags='-X github.com/evcc-io/evcc/server.Version=0.132.1 -X github.com/evcc-io/evcc/server.Commit=64aad0ceb -s -w'"
            ],
            "Environment": [
                "EVCC_NETWORK_PORT=80",
                "EVCC_DATABASE_DSN=/perm/evcc.db"
            ]
        },
        "github.com/gokrazy/breakglass": {
            "CommandLineFlags": [
                "-authorized_keys=/etc/breakglass.authorized_keys",
                "-forward=private-network"
            ],
            "ExtraFilePaths": {
                "/etc/breakglass.authorized_keys": "breakglass.authorized_keys"
            }
        },
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up",
                "--ssh=true"
            ]
        }
    },
    "SerialConsole": "ttyAMA0,115200",
    "InternalCompatibilityFlags": {}
}
```

However SBOM fails:

```
❯ gok -i evcc --parent_dir packaging/gokrazy sbom
2025/01/25 11:46:29 Error: build directory "/Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/tailscale.com/cmd/tailscaled" does not exist in "/Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/github.com/evcc-io/evcc"
Try 'gok -i evcc add tailscale.com/cmd/tailscaled' followed by an update.
Afterwards, your 'gok sbom' command should work: chdir /Users/andig/htdocs/evcc/packaging/gokrazy/evcc/builddir/tailscale.com/cmd/tailscaled: no such file or directory
```

The message is a bit confusing since it's not clear why there should be a tailscaled directory at all? Code is still in https://github.com/evcc-io/evcc/pull/18209

andig, 2025-01-25:
It breaks when using (the modified) `parent_dir`:

gok -i foo new
gok -i foo add tailscale.com/cmd/tailscale
gok -i foo add tailscale.com/cmd/tailscaled
gok -i foo sbom
gok -i foo1 --parent_dir foo1 new
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscale
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscaled
gok -i foo1 --parent_dir foo1 sbom

damdo, 2025-01-25:
I tested the not-ok reproducer, it works for me:

```
[/tmp] $ mkdir foo1
gok -i foo1 --parent_dir foo1 new
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscale
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscaled
gok -i foo1 --parent_dir foo1 sbom
gokrazy instance configuration created in foo1/foo1/config.json
(Use 'gok -i foo1 edit' to edit the configuration interactively.)

Use 'gok -i foo1 add' to add packages to this instance

To deploy this gokrazy instance, see 'gok help overwrite'
2025/01/25 12:02:06 Adding tailscale.com/cmd/tailscale as a (non-local) package to gokrazy instance foo1
2025/01/25 12:02:13 Adding the following package to gokrazy instance "foo1":
  Go package  : tailscale.com/cmd/tailscale
  in Go module: tailscale.com
2025/01/25 12:02:13 Creating gokrazy builddir for module tailscale.com
2025/01/25 12:02:13 Creating go.mod based on upstream go.mod
go: downloading go1.23.1 (darwin/arm64)
2025/01/25 12:02:20 Adding package to gokrazy config
2025/01/25 12:02:20 Adding tailscale.com/cmd/tailscaled as a (non-local) package to gokrazy instance foo1
2025/01/25 12:02:20 Adding the following package to gokrazy instance "foo1":
  Go package  : tailscale.com/cmd/tailscaled
  in Go module: tailscale.com
2025/01/25 12:02:20 Adding require line to existing go.mod
2025/01/25 12:02:20 Adding package to gokrazy config
2025/01/25 12:02:20 ExtraFilePaths of github.com/gokrazy/breakglass: stat foo1/foo1/breakglass.authorized_keys: no such file or directory
[/tmp] $ 
```

andig, 2025-01-25:
@damdo just to be sure- are you using latest gokrazy commit from https://github.com/gokrazy/tools/commit/5762235ac545ccd107e578768f0e67c3d99a9973?

damdo, 2025-01-25:
Oh yes you are right, with the latest gok I get the same error:

```
[/tmp] $ mkdir foo1
gok -i foo1 --parent_dir foo1 new
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscale
gok -i foo1 --parent_dir foo1 add tailscale.com/cmd/tailscaled
gok -i foo1 --parent_dir foo1 sbom
gokrazy instance configuration created in foo1/foo1/config.json
(Use 'gok -i foo1 edit' to edit the configuration interactively.)

Use 'gok -i foo1 add' to add packages to this instance

To deploy this gokrazy instance, see 'gok help overwrite'
2025/01/25 13:07:40 Adding tailscale.com/cmd/tailscale as a (non-local) package to gokrazy instance foo1
2025/01/25 13:07:43 Adding the following package to gokrazy instance "foo1":
  Go package  : tailscale.com/cmd/tailscale
  in Go module: tailscale.com
2025/01/25 13:07:43 Creating gokrazy builddir for module tailscale.com
2025/01/25 13:07:43 Creating go.mod based on upstream go.mod
2025/01/25 13:07:43 Adding package to gokrazy config
2025/01/25 13:07:43 Adding tailscale.com/cmd/tailscaled as a (non-local) package to gokrazy instance foo1
2025/01/25 13:07:43 Adding the following package to gokrazy instance "foo1":
  Go package  : tailscale.com/cmd/tailscaled
  in Go module: tailscale.com
2025/01/25 13:07:43 Adding require line to existing go.mod
2025/01/25 13:07:43 Adding package to gokrazy config
go: finding module for package github.com/gokrazy/firmware
go: finding module for package github.com/gokrazy/rpi-eeprom
go: finding module for package github.com/gokrazy/fbstatus
go: finding module for package github.com/gokrazy/breakglass
go: finding module for package github.com/gokrazy/gokrazy
go: finding module for package github.com/gokrazy/serial-busybox
go: finding module for package github.com/gokrazy/gokrazy/cmd/randomd
go: finding module for package github.com/gokrazy/kernel.rpi
go: finding module for package github.com/gokrazy/hello
go: finding module for package github.com/gokrazy/gokrazy/cmd/heartbeat
go: finding module for package github.com/gokrazy/gokrazy/cmd/dhcp
go: finding module for package github.com/gokrazy/gokrazy/cmd/ntp
go: downloading github.com/gokrazy/gokrazy v0.0.0-20250121184834-730ef75b0f9b
go: downloading github.com/gokrazy/serial-busybox v0.0.0-20250119153030-ac58ba7574e7
go: downloading github.com/gokrazy/gokrazy v0.0.0-20250121184834-730ef75b0f9b
go: downloading github.com/gokrazy/gokrazy v0.0.0-20250121184834-730ef75b0f9b
go: downloading github.com/gokrazy/gokrazy v0.0.0-20250121184834-730ef75b0f9b
go: downloading github.com/gokrazy/gokrazy v0.0.0-20250121184834-730ef75b0f9b
2025/01/25 13:07:56 Error: build directory "/private/tmp/foo1/foo1/builddir/tailscale.com/cmd/tailscale" does not exist in "/private/tmp/foo1/foo1/builddir/github.com/gokrazy/breakglass"
Try 'gok -i foo1 add tailscale.com/cmd/tailscale' followed by an update.
Afterwards, your 'gok sbom' command should work: chdir /private/tmp/foo1/foo1/builddir/tailscale.com/cmd/tailscale: no such file or directory
[/tmp] $
```

stapelberg, 2025-01-25:
Should be fixed with commit https://github.com/gokrazy/tools/commit/70c48e594cf177783248dc4d5afdbcee8ff80bb7, thanks for the reproducer.

As you can tell, few people use relative parent directories.
I’m happy to fix these issues as long as you are willing to report them :)

closed 2025-01-25 by stapelberg

andig, 2025-01-26:
I'll keep it coming as long as I'm allowed to ;)