gokrazy/gokrazy #294: Tailscale: expose port

andig, 2025-02-02:
Bundled tailscale allows exposing SSH:

```json
        "tailscale.com/cmd/tailscale": {
            "CommandLineFlags": [
                "up",
                "--ssh=true"
            ]
        }
```

I'd also like to expose the application port. Usually this would be done with something like

    tailscale serve localhost:80

Is it possible to chain multiple tailscale commands?

andig, 2025-02-02:
The alternative solution would be to interact with the tailscale local client Go API using `lc.SetServeConfig`. I did unfortunately not manage to produce a working forwarding.

stapelberg, 2025-02-04:
> Is it possible to chain multiple tailscale commands?

No, chaining commands in that sense is not possible. gokrazy is meant to manage services (which can either be supervised in a loop or run just once at boot). gokrazy does not aim to provide a sophisticated init system like systemd and instead moves this complexity to the user :) 

Adding the tailscale CLI as a gokrazy package and running it with `up` is already a… creative setup (yes, I realize that this is the setup I recommend in the docs). A cleaner, but less convenient setup would be to set the [DontStart option](https://gokrazy.org/userguide/instance-config/#packagedontstart) so that the CLI is only available for local usage (by humans or other programs), but no command is run automatically.

If you want to do something outside of this model, you need to do it from Go code. So, you could add a program which would then run the two commands in sequence. But, you might not need to add a separate program for this; read on:

> The alternative solution would be to interact with the tailscale local client Go API using `lc.SetServeConfig`. I did unfortunately not manage to produce a working forwarding.

Yeah, I think the best solution if you want to manage tailscale is to either use its API, or, just run the tailscale CLI commands using `exec.Command` from your program? Like so:

```go
// TODO: inspect current configuration to see if any changes are needed
up := exec.Command("/user/tailscale", "up", "--ssh")
up.Stdout = os.Stdout
up.Stderr = os.Stderr
if err := up.Run(); err != nil {
  return fmt.Errorf("%v: %v", up.Args, err)
}
// […] repeat for serve and possibly other commands
```

I hope this helps. Let me know if you can’t make it work.

andig, 2025-02-13:
Found the api way to do this:

```go
	st, err := lc.StatusWithoutPeers(context.Background())
	if err != nil {
		return err
	}
	if st.Self == nil {
		return errors.New("no self node")
	}
	dnsName := strings.TrimSuffix(st.Self.DNSName, ".")

	sc, err := lc.GetServeConfig(context.Background())
	if err != nil {
		return err
	}

	sc.SetWebHandler(&ipn.HTTPHandler{
		Proxy: fmt.Sprintf("http://localhost:%d", port),
	}, dnsName, 443, "/", true)

	return lc.SetServeConfig(context.Background(), sc)
```

This will expose and HTTP service over HTTPS. Note that the `/` mount point is required, empty won't work. To be entirely happy I still need to find a solution to https://github.com/tailscale/tailscale/issues/15002 for obtaining the login url.

stapelberg, 2025-02-22:
> Found the api way to do this:

Great! So does that mean we can close this issue, or is there anything left to be done from the gokrazy side?

andig, 2025-02-22:
We‘re still missing to obtain the TS login url. There seems to be no api way of doing so. But then that‘s as it is…

closed 2025-02-22 by andig