gokrazy/gokrazy #295: Unable to fetch raw log with non-default HttpPort

andig, 2025-02-02:
I couldn't find root cause yet since I can only test on qemu. Fetching raw log doesn't work:

    http://localhost:8888/status?path=%2fgokrazy%2fdhcp

just hangs with 

    "HTTPPort": "8080",

and qemu configured with

   -netdev user,id=net0,hostfwd=tcp::8080-:80,hostfwd=tcp::8022-:22,hostfwd=tcp::8888-:8080

The gokrazy UI itself is available on :8888.

andig, 2025-02-02:
Another small oddity: if I click the dhcp service in the name column, browser url changes to 

    http://localhost:8888/status?path=%2fgokrazy%2fdhcp#%2fgokrazy%2fdhcp

This is the HTML for the link:

    <a href="#%2fgokrazy%2fdhcp">/gokrazy/dhcp</a>

stapelberg, 2025-02-04:
Hey @andig 

> Another small oddity: if I click the dhcp service in the name column, browser url changes to

Yeah, I’m not sure why that was linked. Cleaned up with commit https://github.com/gokrazy/gokrazy/commit/928579311538dbd90052ec6e5b9162a30b83169d

> I couldn't find root cause yet since I can only test on qemu. I'm on
> 
> ```
> http://localhost:8080/status?path=%2fgokrazy%2fdhcp
> ```
> 
> Fetching raw log doesn't work:
> 
> ```
> http://localhost:8080/log?path=%2fgokrazy%2fdhcp&stream=stdout
> ```
> 
> it just hangs.

Thanks for the detailed report. 

The reason why this request is empty is because `/gokrazy/dhcp` does not log any messages to stdout (but it does log to stderr): 

[image]

If you fetch stderr instead of stdout, you should see some lines.

Note that the response is streamed as a Server-sent events (EventSource): https://developer.mozilla.org/en-US/docs/Web/API/Server-sent_events/Using_server-sent_events if you send the `Accept: text/event-stream` header.

closed 2025-02-04 by stapelberg