gokrazy/gokrazy #297: CGO flags for a particular package

fischerman, 2025-02-11:
Hi, I'm having trouble with setting CGO settings. The package that I want to include with gokrazy uses a statically compiled alsa-lib. I can create a static binary like so:
```
CC=aarch64-linux-gnu-gcc GOARCH="arm64" GOOS="linux" CGO_ENABLED=1 CGO_CFLAGS="--sysroot=/home/personal/repo/bandplay/sysroot" CGO_LDFLAGS="--sysroot=/home/personal/repo/bandplay/sysroot" PKG_CONFIG_SYSROOT_DIR="/home/personal/repo/bandplay/sysroot" go build -ldflags="-linkmode external -extldflags=-static" cmd/beep/main.go
```

`/home/personal/repo/bandplay/sysroot` contains the already compiled version of alsa-lib for ARM (while I work on x86). The resulting binary is statically linked and works on the Raspberry. As far as I understand for gokrazy, I cannot specify the environment variables for building this package only. They apply to all packages. This results in the following errors:

```
CC=aarch64-linux-gnu-gcc GOARCH="arm64" GOOS="linux" CGO_ENABLED=1 CGO_CFLAGS="--sysroot=/home/personal/repo/bandplay/sysroot" CGO_LDFLAGS="--sysroot=/home/personal/repo/bandplay/sysroot" PKG_CONFIG_SYSROOT_DIR="/home/personal/repo/bandplay/sysroot" gok overwrite -i bandplay --full /dev/mmcblk0
gokrazy gok g6bec69 on GOARCH=amd64 GOOS=linux

Build target: CGO_ENABLED=1 GOARCH=arm64 GOARCH=arm64 GOOS=linux GOOS=linux
Build timestamp: 2024-11-03T13:19:24+01:00
Loading system CA certificates from /etc/ssl/certs/ca-certificates.crt
Building 5 Go packages:

  github.com/gokrazy/fbstatus

  github.com/gokrazy/hello

  github.com/gokrazy/serial-busybox

  github.com/gokrazy/breakglass
    will be started with command-line flags
      from /home/personal/gokrazy/bandplay/config.json
      last modified: 2024-11-03T13:11:48+01:00 (7m37s ago)

  github.com/fischerman/bandplay/cmd/bandplay
    will be compiled with build flags
      from /home/personal/gokrazy/bandplay/config.json
      last modified: 2024-11-03T13:11:48+01:00 (7m37s ago)

[building (go compiler)]# github.com/hajimehoshi/oto
In file included from <command-line>:
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:10:9: warning: "__STDC_UTF_16__" redefined
   10 | #define __STDC_UTF_16__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:11:9: warning: "__STDC_UTF_32__" redefined
   11 | #define __STDC_UTF_32__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
# github.com/hajimehoshi/oto
In file included from <command-line>:
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:10:9: warning: "__STDC_UTF_16__" redefined
   10 | #define __STDC_UTF_16__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:11:9: warning: "__STDC_UTF_32__" redefined
   11 | #define __STDC_UTF_32__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
# github.com/hajimehoshi/oto
In file included from <command-line>:
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:10:9: warning: "__STDC_UTF_16__" redefined
   10 | #define __STDC_UTF_16__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
../../repo/bandplay/sysroot/usr/include/stdc-predef.h:11:9: warning: "__STDC_UTF_32__" redefined
   11 | #define __STDC_UTF_32__ 1
      |         ^~~~~~~~~~~~~~~
<built-in>: note: this is the location of the previous definition
[done] in 7.33s                        

Including extra files for Go packages:

  github.com/gokrazy/breakglass
    will include extra files in the root file system
      from /home/personal/gokrazy/bandplay/breakglass.authorized_keys
      last modified: 2024-11-03T12:24:15+01:00 (55m18s ago)

Including loadable kernel modules from:
/home/personal/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20240802144420-fd363f1112d3/lib/modules
2024/11/03 13:19:37 CheckDir(/home/personal/repo/bandplay): /home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_connmark.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_CONNMARK.h" and "sysroot/usr/include/linux/netfilter/xt_connmark.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_dscp.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_DSCP.h" and "sysroot/usr/include/linux/netfilter/xt_dscp.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_mark.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_MARK.h" and "sysroot/usr/include/linux/netfilter/xt_mark.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_rateest.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_RATEEST.h" and "sysroot/usr/include/linux/netfilter/xt_rateest.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_tcpmss.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_TCPMSS.h" and "sysroot/usr/include/linux/netfilter/xt_tcpmss.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter_ipv4/ipt_ecn.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter_ipv4/ipt_ECN.h" and "sysroot/usr/include/linux/netfilter_ipv4/ipt_ecn.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter_ipv4/ipt_ttl.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter_ipv4/ipt_TTL.h" and "sysroot/usr/include/linux/netfilter_ipv4/ipt_ttl.h"
/home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter_ipv6/ip6t_hl.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter_ipv6/ip6t_HL.h" and "sysroot/usr/include/linux/netfilter_ipv6/ip6t_hl.h"
```

The warnings from oto can be ignored. Looks like the sysroot is considered while including kernel modules and conflicts with ... something?! Any idea how I can get around this?

stapelberg, 2025-02-12:
Hey @fischerman 

> As far as I understand for gokrazy, I cannot specify the environment variables for building this package only. They apply to all packages.

That’s correct. We could easily add a `GoBuildEnvironment` option, though. Which packages do you need to compile with conflicting options?

---

Let me try and give you some pointers for debugging the build failure:

> > Including loadable kernel modules from:
/home/personal/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20240802144420-fd363f1112d3/lib/modules
2024/11/03 13:19:37 CheckDir(/home/personal/repo/bandplay): /home/personal/repo/bandplay/sysroot/usr/include/linux/netfilter/xt_connmark.h: case-insensitive file name collision: "sysroot/usr/include/linux/netfilter/xt_CONNMARK.h" and "sysroot/usr/include/linux/netfilter/xt_connmark.h"
> 
> Looks like the sysroot is considered while including kernel modules 

No, I think the kernel modules were [successfully discovered](https://github.com/gokrazy/tools/blob/30b9fe0c81f88b4f272e7b4023b6ce5f527ebebd/internal/packer/packer.go#L1313-L1324), and it’s likely the next step that’s failing: [Generating the SBOM](https://github.com/gokrazy/tools/blob/30b9fe0c81f88b4f272e7b4023b6ce5f527ebebd/internal/packer/packer.go#L1431).

You can confirm this by running `gok sbom` instead of `gok overwrite`. 

> and conflicts with ... something?! 

gokrazy records the hash of the go.mod file (like recorded in a go.sum file) of your instance’s packages in its SBOM.

We use the same hash implementation as the Go tool itself (so that we know for sure it’s compatible with code that can be distributed as a Go module), and that algorithm has a couple of restrictions: https://go.dev/ref/mod#zip-path-size-constraints. The one you are running into is:

> No two files within a zip file may have paths equal under Unicode case-folding (see strings.EqualFold). This ensures that zip files can be extracted on case-insensitive file systems without collisions.

So operating systems like macOS with their case insensitive file names are causing build failures on your Linux machine… :) 

One might think that maybe we could relax the requirements and stick to the same hashing algorithm, but that’s not the case: https://github.com/gokrazy/tools/commit/80a59f115332a41206487afc6038beaaed48ce70. One thing we might be able to do is switch to a custom hashing algorithm altogether, but in your case, I think the best fix follows below:

> Any idea how I can get around this?

I suspect you are using a `replace` directive in the `builddir/github.com/fischerman/bandplay/cmd/bandplay/go.mod` file so that `gok sbom` starts reading from `/home/personal/repo/bandplay` on your local disk (as opposed to the go module cache) and that same directory contains the sysroot, too.

Traversing a full kernel source on every `gok` build will be annoyingly slow (if it worked at all), so you should move the sysroot directory outside of your Go program repositories.

fischerman, 2025-02-13:
You are spot on. `gok sbom` with no environment variables fails with the same error. Once I move sysroot out of the module it works.

I was misled by the kernel modules because the log looked like it belonged to the same step. That means I don't need a `GoBuildEnvironment`. It would only be a convenience (to not specify the environment variables every time I call gok).

closed 2025-02-13 by fischerman

stapelberg, 2025-02-21:
FYI, with commit https://github.com/gokrazy/tools/commit/703a8605028963d13a9d00815ccedfae13f0ce6d, this restriction should be lifted