gokrazy/gokrazy #299: Build fails on unrelated files

andig, 2025-02-12:
My local module directory contains some unrelated files, in this case from testing. This confuses the gok build:

```
Including loadable kernel modules from:
/Users/andig/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20250109091720-7437caa083f3/lib/modules
2025/02/12 17:32:09 CheckDir(/Users/andig/htdocs/evcc): /Users/andig/htdocs/evcc/test-results/heating-loadpoint-change-limit-in-1°-steps-chromium/trace.zip: malformed file path "test-results/heating-loadpoint-change-limit-in-1°-steps-chromium/trace.zip": invalid char '°'
make: *** [gok-vm] Error 1
```

It is not clear, why these files should be part of the build?

stapelberg, 2025-02-12:
What a coincidence. https://github.com/gokrazy/gokrazy/issues/297 is almost the same issue. Please read my reply there from just a minute ago and then report back if there are any questions left.

Maybe the easiest fix is to just change the test-result file names to be valid in a Go module?

stapelberg, 2025-02-12:
> They are just in an arbitrary folder under the module root. `go build` is happy with them being present.

Yeah, you can *build* them, but you can’t *distribute* them as a Go module.

> Do you mean that it traverses the entire module packages and treats each folder as if it was a used package?

Yes. `gok` can’t know which parts of your directory end up influencing the build, so it hashes everything. This is the same way Go modules work (they record a hash for all files inside the module).

The SBOM is used for updating (with GUS, the gokrazy update service) to figure out when an update is needed or not. So if we do not capture all files, that would break updating: you could change something in your working copy and gokrazy would claim the instance hasn’t changed and is already up-to-date! Not good. 

But maybe one thing we could do is change the SBOM generation to hash the artifacts that gokrazy uses (i.e. build first, then generate SBOM) instead of hashing the sources that result in the artifacts. @damdo Any thoughts about this approach? Should work just as well for updating, right?

andig, 2025-02-12:
That might even be faster?

damdo, 2025-02-21:
> But maybe one thing we could do is change the SBOM generation to hash the artifacts that gokrazy uses (i.e. build first, then generate SBOM) instead of hashing the sources that result in the artifacts. @damdo Any thoughts about this approach? Should work just as well for updating, right?

@stapelberg hm then `gok sbom` would always trigger a build and then compute at the end?

stapelberg, 2025-02-21:
> > But maybe one thing we could do is change the SBOM generation to hash the artifacts that gokrazy uses (i.e. build first, then generate SBOM) instead of hashing the sources that result in the artifacts. [@damdo](https://github.com/damdo) Any thoughts about this approach? Should work just as well for updating, right?
> 
> [@stapelberg](https://github.com/stapelberg) hm then `gok sbom` would always trigger a build and then compute the hash at the end?

Yes. I just implemented this and it seems to work well. Will push in a second, please let me know if this broke something :)

closed 2025-02-21 by stapelberg

andig, 2025-02-22:
@stapelberg would you expect this change to speed up the sbom step (the virus scanner issue)? Seems its one less pass over the source files?

stapelberg, 2025-02-22:
Maybe? On my system, the command is fast no matter what (before and after my change). Give it a try and see if it makes a difference for you.