gokrazy/gokrazy #300: Ignore podman interface prefix in PrivateInterfaceAddrs

lordwelch, 2025-02-16:
PrivateInterfaceAddrs collects all the private interfaces and is used by some services to listen on all the local interfaces.
This can cause issues, or at the very least useless log messages like this in services that use it to listen on private interfaces (logs from [rtr7/router7/cmd/dnsd]):
```plain
2025/02/16 15:08:55 listener for "10.89.0.1" died: listen udp 10.89.0.1:53: bind: address already in use
2025/02/16 15:08:55 listener for "10.89.0.1" died: listen tcp 10.89.0.1:53: bind: address already in use
```

Since podman by default will create a bridge interface with the prefix 'podman' I think ignoring them should be an easy default to add.

[rtr7/router7/cmd/dnsd]: https://github.com/rtr7/router7/tree/master/cmd/dnsd

stapelberg, 2025-02-18:
It’s probably possible, yes, but I am not sure if ignoring podman addresses really is the right thing to do.

The 10/8 network is a private network, so I would actually expect daemons that provide internal services (like DNS resolution) to listen on such IP addresses.

lordwelch, 2025-02-23:
After investigating more, it probably isn't a good default to ignore the podman interfaces. I had some wrong assumptions about how the network stack worked.

The vast majority of services default to listening to all address (0.0.0.0) which which would still listen on the podman interfaces on a normal linux machine, the difference is that gokrazy binds to all the available private addresses directly which is a fair choice (as otherwise a firewall would probably need to be setup which is more complicated) and just happens to conflict with podman starting it's own DNS server to handle container name resolution on networks with dns_enabled set to true (which also only happens when running podman as root, rootless uses namespaces). Normal services listening on 0.0.0.0 wouldn't have that conflict as the kernel doesn't see binding to a specific 10/8 address as a conflict it just gives it priority.

Thanks for considering it anyways

closed 2025-02-23 by lordwelch