gokrazy/gokrazy #305: web UI often not (re)loading over tailscale

mpl, 2025-03-06:
So I have this big blocker which happens fairly often.

As title says, sometimes the web UI won't load, or reload, if it is being served over tailscale. I see the problem either on the main page with all the services, or for any given individual service page.
Sometimes it succeeds immediately, sometimes it succeeds after several minutes, and sometimes it just never succeeds. Forcing it to reload does not seem to help.
At first I thought it was browser related, as it didn't seem to happen on firefox (VS chrome), but even on firefox I'm  seeing the problem too now.

If I switch to accessing it over its other IP (atm on my LAN WIFI), then no problem. But that's not a viable alternative, as the devices will eventually be on a public, most likely changing, IP.

No visible clue in the developer tools.

Any idea on how to debug that further, or of a possible work-around please?

stapelberg, 2025-03-06:
I access many of my devices via Tailscale and haven’t noticed issues like what you describe.

To debug this further, I would rule out the browser by right-clicking the request in the network inspector devtools tab, then selecting “export as cURL” and running the request on the terminal. Then, cross-reference the tailscale logs of your local machine and the gokrazy device to see if there are any hints in there.

mpl, 2025-03-06:
I didn't know the "export as cURL" functionality, thanks.
So yeah, no problem when fetching directly with curl.

On the other hand, new symptom: it's now also happening directly over my WIFI, directly by LAN IP.
So it would actually have nothing to do with tailscale after all. So weird.

stapelberg, 2025-03-07:
OK, perhaps the issue is related to the EventSource streams we use for service stdout/stderr streaming: https://github.com/gokrazy/gokrazy/blob/9874e1e6d72912168e05ebf858345d41309b38b9/internal/assets/status.tmpl#L79

(You could try commenting out the code to verify.)

There is a 6 connections per host limit when you don’t use HTTPS: https://developer.mozilla.org/en-US/docs/Web/API/EventSource, so perhaps you hit that with enough open tabs or something like that.

In my experience, pressing Escape to cancel a page load and then navigating to the page immediately again (sometimes repeatedly) seems the quickest way to get things unstuck.

Let me know if this makes sense for the symptoms you are seeing.

mpl, 2025-03-07:
Ah, interesting.
But the EventSource is not involved when serving the "main" page with all the services, is it?
I'm asking because I see the problem even on this main page.

And yes, I've had a limited success so far with cancelling the page load, and retrying, etc., but it does not "fix" the problem very often.

I've also been looking into other browsers in the meantime, as a temporary work-around. Seems like there's no problem with Lynx, but text-based is a bit rough ;)

stapelberg, 2025-03-07:
> But the EventSource is not involved when serving the "main" page with all the services, is it? I'm asking because I see the problem even on this main page.

Correct, there is no EventSource on the main page. Are you saying you can reproduce the problem with *only main pages* open and ever opened? I’m just wondering because EventSource connections will stay alive for a while, even after navigating away from the page.

mpl, 2025-03-07:
Hah! good call, thanks.
If I open only the main page I can open it in >10 tabs without any issue.
If I open tabs of services, at the 3rd of 4th one, I can start to notice the failure. And as soon as I close a couple of them, it seems to calm down.
But If open enough service tabs to make things fail, then if I try to reload the main page, it fails to reloads as well, which is why I thought the problem was not related to a service page in particular. :)

I'll keep on monitoring, but I think you found the issue with EventSource indeed.

I suppose I could eventually add some let's encrypt to the mix to have HTTPS/HTTP2 everywhere and overcome the limitation.

mpl, 2025-03-19:
EventSource was almost definitely the reason since I haven't had anymore problems now that I've been careful with the number fo tabs opened.

stapelberg, 2025-04-13:
FYI, I have pushed a few commits that hopefully help the EventSource situation. Let me know if you still see weird behavior after upgrading the `github.com/gokrazy/gokrazy` package to commit e01459ea628ba1ebb19ea28f0352e52fd708b8ff or newer.

mpl, 2025-04-14:
oh interesting, thanks.
But does that mean that if I'm not watching the tab, then I come back to it, then the stuff that happened in the meantime does not end up being displayed?
Hmm, maybe I'm not explaining it well enough. I'll try it out to see for myself :)