gokrazy/gokrazy #307: eMMC support on raspberry pi 5

mpl, 2025-03-19:
Hello,

I have found eMMC mentioned at https://gokrazy.org/platforms/ , and in https://github.com/gokrazy/gokrazy/issues/270 but I am not sure whether the limitation is only for Pine64 Rock64 , or if it is a general limitation for all boards?

In other words, I was planning on using gokrazy on a raspberry pi 5 compute module, where one is supposed to install the system on eMMC rather than on SD card. Is this somehow possible (even if it involves extra work-arounds), or should I give up on gokrazy for that board?

thanks

stapelberg, 2025-03-19:
eMMC storage should work fine, I think? Things might work out of the box, just give it a try.

(I have a Compute Module 5 here that I have not yet found time to install, but I am planning on making it work eventually. If you report specific issues, I might be able to take a look.)

mpl, 2025-03-19:
Ah sorry, I didn't make myself clear.

What I mean is: for a normal pi5, for the initial provisioning, I'm running "gok overwrite --full /dev/blabla" from my laptop, to write the system onto the SD card, right?

So given that an eMMC is not removable (I think?) from the pi5, how would I go about that initial provisioning in the case of a compute module?

I'm not familiar with the compute module or eMMC, so it's entirely I'm missing something here though.

stapelberg, 2025-03-19:
Ah. I would recommend to write gokrazy to a USB memory stick and boot the compute module from that, then use `dd` to copy the memory stick contents over to the eMMC storage you want to use.

mpl, 2025-03-19:
Thanks, I'll try that out.

mpl, 2025-04-08:
So I've finally received my CM5, and the IO Board to mount it on, so I can report more on the issue.

First of all, I don't think the CM5 can boot from USB or SDCard, or at least not by default. I did try booting gokrazy from either, and it didn't seem to work. I think it can only boot from SDCard with models without an eMMC (the "L" like "Light" ones).
But I think it might still be possible by changing the boot order in the bootloader, as explained at https://www.raspberrypi.com/documentation/computers/compute-module.html#compute-module-eeprom-bootloader although I did not try.

Anyway, maybe I'm wrong on that specific point, but no matter, because I still succeeded by slightly adjusting your proposed method. :)

I started as you suggested, which is as usual, i.e. writing the first gokrazy install to SDCard or USB disk.
Then I followed the instruction at https://www.raspberrypi.com/documentation/computers/compute-module.html#flash-compute-module-emmc , which basically explains how to turn the eMMC of the CM5 into a mass storage device, as far as your host (your laptop or whatever else) is concerned. Then one can usually write an OS image with dd or a fancy GUI onto said mass storage device. But in our case, I used dd to copy the contents of my gokrazy USB stick onto the eMMC. So from the host perspective it's copying from one removable storage device, onto another.

One caveat though, is since the gokrazy install by default uses the whole device to make the partitions, one needs to have a removable storage smaller than the eMMC, otherwise you're going to end up copying a partition scheme that does not fit on the eMMC.

But other than that, not difficult overall so far.

I guess the first step to make all that a bit less awkward would be to be able to have gokrazy in an .img instead of on a removable storage.

And just as I'm writing that, I'm discovering that gok overwrite --full can take a .img path as argument, and I'm guessing --target_storage_bytes is what will help me make it fit the size of the eMMC in advance. So maybe my sentence above can already be scratched. I'm going to try that right away then. :)

mpl, 2025-04-08:
In any case, I don't know with a custom board, but with the CM5 on the official IO board, yes, gokrazy can be installed onto the eMMC, and runs. So this issue can be closed as far as I'm concerned.

mpl, 2025-04-08:
Ok, one thing that does not seem to work: updates.

I've tried both with selfupdate+GUS or with direct update with ```gok update``` , and in both cases everything seems to be going fine, except after the automatic reboot, the device is still on the old revision.

Not sure yet how to debug that.
Any advice please?

stapelberg, 2025-04-13:
Hey @mpl, glad you got a bit further with your installation :)

Can you share the full output you’re getting when you run `gok update`? That will help me ask some useful follow-up questions for further debugging.

I assume you have some sort of interactive access to the device now? A serial console, or access via breakglass perhaps? What is the output of `dmesg`?

mpl, 2025-04-13:
To my great shame, I must admit I fixed the problem (or rather it fixed itself) without knowing how.

For a couple of days I tried different things, like going back to a simpler config.json, enabling -testboot, etc, and nothing seemed to work, neither after selfupdate or after a manual update.
I had started reading the updating code and the booting code to understand the whole process, I had mounted the non-active partition to try and examine what was on it, etc.
Then on the third day (I had turned off the pi for the night), when I booted it then the update was finally live on it.
I wonder if it was some sort of caching problem that would only go away after actually completely shutting off the device.
I keep track of everything I do, but this time I am puzzled as to what happened and how it got fixed.

I suppose I could try and go back to a "broken" state by starting everything from scratch, but I'd rather not, as I still have many other things to explore. (there's at least one bug - unrelated to CM5 - I haven't reported yet, but this one was easy to work-around, and I'll report it + send a PR a bit later).

I'll have more CM5s to setup in the near future, so I'll be sure to report again if I hit the same problem of course.

stapelberg, 2025-04-14:
Alright, thanks for letting me know. I’ll keep some notes when I’ll set up my CM5, too.

closed 2025-04-14 by stapelberg

ben-auo, 2025-07-04:
FYI, for those looking for an IMG/ISO builder it took me a bit to realize gok supports this natively:

`gok -i test overwrite --full test.img --target_storage_bytes 1258299392 && gzip -k test.img`