gokrazy/gokrazy #310: sound support?

mpl, 2025-04-17:
Hi,

I have some Go code that I was successfully using (on BalenaOS) on rpi 5 to record sound over a USB microphone.
It depends on `github.com/gordonklaus/portaudio` , so behind the scenes it is some CGO, and some ALSA dev libs are needed.

Anyway, setting aside my code for a moment, just for the sake of debugging, I managed to get some of the alsa utils (like aplay, arecord) running on gokrazy through the freeze + LD_LIBRARY_PATH=$PWD ./ld-linux-aarch64.so.1 method.
And then:
```
# LD_LIBRARY_PATH=$PWD ./ld-linux-aarch64.so.1 /usr/bin/arecord -l
arecord: device_list:274: no soundcards found...arecord -l
```

And from I've seen here and there in docs, it seems to me that the `snd-bcm2835` module should be loaded for the onboard soundcard to be working, no? And yet, this is what I get with `lsmod`:
```
brcmfmac_wcc 12288 0 - Live 0xffffa00018037000
brcmfmac 282624 1 brcmfmac_wcc, Live 0xffffa00017fdd000
brcmutil 16384 1 brcmfmac, Live 0xffffa00017fd6000
pwm_fan 20480 0 - Live 0xffffa00017fcd000
```
And I when I try to `modprobe` it, it isn't found.

I've also added
```dtparam=audio=on```
to config.txt on the boot partition, and rebooted, but it didn't seem to change anything.

What do you think? Is sound supposed to be working out of the box in gokrazy?
If not, aside from all the ALSA stuff, is there anything I need to do at the module level to get a working soundcard?

stapelberg, 2025-04-18:
Hey @mpl and happy easter!

In general, I don’t have any audio-related use-cases, so I don’t know for sure either…

But I do know that @anisse has built https://github.com/anisse/beatbox, which seems to use ALSA (via the https://github.com/anisse/alsa package) and is running on the Pi 3 with the gokrazy/kernel package. So, in general, some level of working ALSA should be possible (or at least was possible in the past).

Regarding the kernel: gokrazy originally did not have support for loadable kernel modules. Instead, we compiled almost everything into the kernel. Only when adding support for WiFi did we need to make loadable kernel modules (and firmware loading) work, but that’s the reason why our kernel still compiles in many modules instead of compiling them as module.

Therefore, the correct check is to not only look at loadable moudles, but instead to look at the kernel config via `/proc/config.gz`. The gokrazy/kernel kernel has the bcm2835 sound driver built-in:

```
/tmp/breakglass3319870921 # uname -a
Linux dr 6.13.4 #1 SMP PREEMPT Wed Mar  1 20:57:29 UTC 2017 aarch64 GNU/Linux

/tmp/breakglass3319870921 # gunzip -c /proc/config.gz  | grep SND_BCM
CONFIG_SND_BCM2835_SOC_I2S=y
# CONFIG_SND_BCM63XX_I2S_WHISTLER is not set
CONFIG_SND_BCM2835=y
```

The gokrazy/kernel.rpi kernel does not seem to have anything sound-related enabled, though:

```
/tmp/breakglass1484950224 # uname -a
Linux scan2drive 6.6.74-v8 #1 SMP PREEMPT Wed Mar  1 20:57:29 UTC 2017 aarch64 GNU/Linux
/tmp/breakglass1484950224 # gunzip -c /proc/config.gz  | grep SND_
/tmp/breakglass1484950224 # 
```

Maybe the kernel.rpi defaults do not include sound support, or use a different module? We might need to add some options to https://github.com/gokrazy/kernel.rpi/blob/main/_build/config.addendum.txt

I would have suggested switching to the gokrazy/kernel kernel to see if that makes a difference, but upstream Linux doesn’t have support for the Raspberry Pi 5 yet :(

mpl, 2025-04-18:
> Hey [@mpl](https://github.com/mpl) and happy easter!

Happy easter to you too

> In general, I don’t have any audio-related use-cases, so I don’t know for sure either…
>
> But I do know that [@anisse](https://github.com/anisse) has built https://github.com/anisse/beatbox, which [...]

Excellent, I will check it out, thanks.

> Regarding the kernel: gokrazy originally did not have support for loadable kernel modules. Instead, we compiled almost everything into the kernel. Only when adding support for WiFi did we need to [...]

Sorry, I'm not familiar with kernels on gokrazy. Of these two outputs you're showing, why and how are they using a different kernel? Are these two both running on raspberry pis?

> Maybe the kernel.rpi defaults do not include sound support, or use a different module? We might need to add some options to [...]

stapelberg, 2025-04-18:
> Sorry, I'm not familiar with kernels on gokrazy. Of these two outputs you're showing, why and how are they using a different kernel? Are these two both running on raspberry pis?

Yes, both are running on a Raspberry Pi.

See https://gokrazy.org/platforms/#gokrazy-kernel-repository-map for an overview of which kernels are available and why.

You can select the kernel by changing the KernelPackage setting: https://gokrazy.org/userguide/instance-config/#kernelpackage

mpl, 2025-04-18:
> Yes, both are running on a Raspberry Pi.
>
> See https://gokrazy.org/platforms/#gokrazy-kernel-repository-map for an overview of which kernels are available and why.
>
> You can select the kernel by [...]

Ok, so you're saying my best bet is to follow the instruction at https://github.com/gokrazy/kernel.rpi/tree/main , to try and build my own kernel.rpi, but with sound included, right?
And then use that new kernel with the KernelPackage directive?

stapelberg, 2025-04-18:
> Ok, so you're saying my best bet is to follow the instruction at https://github.com/gokrazy/kernel.rpi/tree/main , to try and build my own kernel.rpi, but with sound included, right?

If you have a Raspberry Pi 4 or older, trying `github.com/gokrazy/kernel` as `KernelPackage` would be even easier. 

But if you only have a Raspberry Pi 5, then yes.

> And then use that new kernel with the KernelPackage directive?

No, you would still specify the same KernelPackage (`github.com/gokrazy/kernel.rpi`), but use the [replace directive](https://gokrazy.org/development/modules/#building-local-code-the-replace-directive) to use your local working copy of the kernel (just like with any other Go module).

mpl, 2025-04-18:
> If you have a Raspberry Pi 4 or older, trying `github.com/gokrazy/kernel` as `KernelPackage` would be even easier.
>
> But if you only have a Raspberry Pi 5, then yes.

Yeah I'm only pi5 :)

> No, you would still specify the same KernelPackage (`github.com/gokrazy/kernel.rpi`), but use the [replace [...]

ah yeah, the benefits of everything is Go all the way :)

Comparing with Anisse's kernel config at https://github.com/anisse/beatbox-kernel/blob/master/beatbox-config , I think it might simply be that we're missing `CONFIG_SND=y` in https://github.com/gokrazy/kernel.rpi/blob/main/_build/config.addendum.txt 
I'll try it out.

mpl, 2025-04-18:
Damn, the build fails without much of a clue for me:

```
  DTCO    arch/arm64/boot/dts/overlays/i2c4.dtbo
  CC      init/main.o
Illegal instruction
make[3]: *** [scripts/Makefile.build:243: init/main.o] Error 132
make[2]: *** [scripts/Makefile.build:480: init] Error 2
make[1]: *** [/usr/src/linux-stable_20250127/Makefile:1921: .] Error 2
make[1]: *** Waiting for unfinished jobs....
  DTC     arch/arm64/boot/dts/qcom/sc7180-trogdor-homestar-r3.dtb
  DTCO    arch/arm64/boot/dts/overlays/i2c5.dtbo
  DTCO    arch/arm64/boot/dts/overlays/i2c6.dtbo
...
  DTC     arch/arm64/boot/dts/overlays/overlay_map.dtb
  DTCO    arch/arm64/boot/dts/overlays/ws2812-pio.dtbo
  DTC     arch/arm64/boot/dts/overlays/hat_map.dtb
make: *** [Makefile:234: __sub-make] Error 2
2025/04/18 15:31:18 make: exit status 2
2025/04/18 17:31:19 docker run: exit status 1 (cmd: [/Applications/Docker.app/Contents/Resources/bin/docker run --platform=linux/amd64 --volume /Users/mpl/src/github.com/gokrazy/kernel.rpi/_build:/tmp/buildresult:Z --rm gokr-rebuild-kernel -cross=arm64 -flavor=raspberrypi https://github.com/raspberrypi/linux/archive/refs/tags/stable_20250127.tar.gz])
```

btw, I'll file an issue about it later if you want, but I think this instruction:
```
GOBIN=$PWD/_build go install github.com/gokrazy/autoupdate/cmd/gokr-rebuild-kernel@latest
cd _build && ./gokr-rebuild-kernel -cross=arm64 -flavor=raspberrypi
```
cannot work as is on mac.
Because the first instruction will build a darwin binary.
And then when the Dockerfile does:
```
COPY gokr-rebuild-kernel /usr/bin/gokr-rebuild-kernel
```
you'll end up with a darwin binary within your docker image, which of course won't work when executed.
But that was easy enough to work around.

mpl, 2025-04-18:
I've also tried reverting to 7437caa083f3b04d56a1795f40bcb6b5becc3e34 (kernel 20241008), but it also fails to build in various places just with
```
Illegal instruction
aarch64-linux-gnu-gcc: internal compiler error: Illegal instruction signal
```

I wonder if it could be my docker desktop setup. I'll update it and give it more memory.

mpl, 2025-04-23:
ouch, now hitting that (among other things):
```

anisse, 2025-04-24:
> But I do know that [@anisse](https://github.com/anisse) has built https://github.com/anisse/beatbox, which seems to use ALSA (via the https://github.com/anisse/alsa package) and is running on the Pi [...]

Hello, please note that my implementation might be suboptimal, but it's simple enough to be read in one sitting. Something like https://github.com/yobert/alsa might be more featureful. You can try its record binary, which will remove the need for any group file or libc.

mpl, 2025-04-30:
> Hello, please note that my implementation might be suboptimal, but it's simple enough to be read in one sitting. Something like https://github.com/yobert/alsa might be more featureful. You can try [...]

Hey,

Thanks for the tip.
I gave github.com/yobert/alsa a try, and I could indeed record from gokrazy with it.

However, it seems you can only do direct hardware access with it (the "hw" device). Common virtual devices (plugins), such as the plughw device don't seem to be accessible with it. And I don't think that's a gokrazy or alsa setup/configuration problem, because the plughw device does show up (and is usable) with e.g. arecord on gokrazy.

So basic things like sample rate conversion would have to done by hand, instead of being able to rely on plughw to do them. That plus the fact that yobert/alsa does not seem super maintained/active makes me a bit hesitant.

I might bite the bullet and try to keep on using github.com/gordonklaus/portaudio , which means to keep on trying with the full alsa libs + CGO setup. Which is also a pain.

By the way, I think the part about the audio group not present was just a warning, and not fatal. And the actual failure later on was that my code wants to use a 16000 sample rate, whereas the hardware only supports 44100. And since plughw is not there to do the conversion on the fly, it all goes boom.

anisse, 2025-04-30:
> So basic things like sample rate conversion would have to done by hand, instead of being able to rely on plughw to do them. 

Yes, that's the magic of alsa-lib, it handles format conversion, resampling (and mixing?) for you. Pulseaudio or Pipewire can also do this for you. Portaudio will use libasound (alsa-lib) so it hides the problem. Note that this is not specific to Go, other languages that don't use alsa-lib directly face the same issue (i.e, bring your own resampling, format conversion), unless your audio daemon hides that for you.

mpl, 2025-05-02:
Just an an FYI, I've just gotten to a state that is satisfactory for me (so far), all done in pure Go. :-)

Using `github.com/yobert/alsa` for raw capturing, and then `github.com/zeozeozeo/gomplerate` to convert from 44100Hz to 16000Hz. (In case anyone's wondering, I need 16000, because I then feed the audio to a speech to text service, which requires this sample rate).

@stapelberg What should we do about this issue? Do you want to keep it open for refining the bare minimum we need in the kernel to get sound recording working in gokrazy out of the box?

stapelberg, 2025-05-03:
Can you share what the setup requires that you ended up with? Ideally with code and setup steps.

That would make it easy for me to say which steps we could easily do in gokrazy.

Thanks

closed 2025-05-06 by stapelberg

mpl, 2025-05-06:
> Perfect! Thanks for testing, I’ll close the issue then.

And thanks for the quick kernel patching!

> If you wanted to share your project in a blog post, I’d be happy to link to it from the gokrazy showcase: https://gokrazy.org/packages/showcase/

Cool. At the moment I'm still testing as much as I can, to know whether to go with Gokrazy or something else (at least one alternative would be balenaOS, but I do have griefs with it), given it's for an actual project at work. I definitely would prefer to keep going with Gokrazy but I must be wary of not succumbing to the sunk cost fallacy and the coolness factor. ;)

It's all going to be open-source eventually, but not before the whole project is over. I can't share too much detail for now, but it's for a conservation project (and hopefully others later) at arribada.org

But yeah, maybe I can share parts of it in a blog post before the whole thing is over. And as soon as we're sure we're going with gokrazy, you can expect more issues/PRs from me too ;)