gokrazy/gokrazy #312: Feature request: add kernel commandline arguments

leo-baltus, 2025-05-15:
Trying out gokrazy on an Intel Nuc with an usb stick did not seem to work because the stick reported its partitions too late, I guess. Because after adding the `rootwait` kernel argument gokrazy could finally boot.

Now in order to make this permanent I would assume some configuration option for this but there is none as far as I can tell. I think there are more use cases where additional kernel arguments would be handy.

stapelberg, 2025-05-20:
After `go install github.com/gokrazy/tools/cmd/gok@fc2e1562e828`, I can adjust `config.json` like so:

```diff
diff --git i/scan2drive/config.json w/scan2drive/config.json
index 599baea..03f78b3 100644
--- i/scan2drive/config.json
+++ w/scan2drive/config.json
@@ -92,6 +92,10 @@
     },
     "SerialConsole": "serial0,115200",
     "KernelPackage": "github.com/gokrazy/kernel.rpi",
+    "KernelExtraArgs": [
+	"gokrazy.rocks=true",
+	"like=definitely"
+    ],
     "FirmwarePackage": "github.com/gokrazy/firmware",
     "EEPROMPackage": "github.com/gokrazy/rpi-eeprom",
     "InternalCompatibilityFlags": {}
```

…and see the args in the kernel command line after `gok update`:

`
/tmp/breakglass851441313 # cat /proc/cmdline
reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave   vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=tty1 console=ttyAMA10,115200 root=PARTUUID=60c24cc1-f3f9-427a-8199-7928a6c20001/PARTNROFF=1 init=/gokrazy/init rootwait panic=10 oops=panic gokrazy.rocks=true like=definitely
`

stapelberg, 2025-05-20:
Now documented at https://gokrazy.org/userguide/instance-config/#kernelextraargs

closed 2025-05-20 by stapelberg

damdo, 2025-05-21:
Thanks @stapelberg !