gokrazy/gokrazy #317: 'tailscale down' persists across reboots

mpl, 2025-05-22:
Steps to reproduce:

1) Have a gokrazy instance setup with functional tailscale
here's my config:
```
		"tailscale.com/cmd/tailscale": {
			"CommandLineFlags": [
				"up",
				"--auth-key=file:/etc/ts.key"
			],
			"ExtraFilePaths": {
				"/etc/ts.key": "ts.key"
			}
		}
```
2) breaglass onto instance
3) run '/user/tailscale down'
As expected, instance is not reachable anymore over tailscale
4) reboot (soft, or cut the power, same result)
Unexpectedly, instance is still not reachable over tailscale

5) Cheat, and reach web UI over its non-tailscale IP (which might not be possible if your device is already in "production", hence the problem), and force "Run once" on tailscale service, to fix the situation.

So maybe I'm missing something, but the reason I'm surprised is since the tailscale service runs '/user/tailscale up' on startup, I would expect the device to be reachable over tailscale after the reboot, even if I manually did 'tailscale down' before the reboot, no?

If you think that is the expected behaviour, and you don't consider it a bug, what do you recommend as a method to take tailscale down temporarily (because I want to play with the nftables without interference from tailscale), but to be sure that it's going to be up again when I reboot the device?

btw, I realise even if it is an unexpected behaviour, it might not be a core gokrazy bug per se. It's rather a documentation/configuration issue I suppose?

stapelberg, 2025-05-22:
I don’t have much time to look into this now, so here’s a quick reply:

I have never run `tailscale down` before, but all other tailscale actions are persistent, so I would not be surprised by the behavior you are describing.

Personally, I just stop tailscaled via the gokrazy web interface if I want to stop tailscale.

mpl, 2025-05-22:
> I don’t have much time to look into this now, so here’s a quick reply:
> 
> I have never run `tailscale down` before, but all other tailscale actions are persistent, so I would not be surprised by the behavior you are describing.

ok, interesting.

> Personally, I just stop tailscaled via the gokrazy web interface if I want to stop tailscale.

ah the whole of tailscaled instead I see, thanks.

Btw, feel free to move this as a discussion too. As I said, at worst I think it's more of configuration/documentation issue.

stapelberg, 2025-05-29:
> I have never run `tailscale down` before, but all other tailscale actions are persistent, so I would not be surprised by the behavior you are describing.

I just tested the behavior on NixOS 25.05: After `tailscale down`, the node does not come back after a reboot. Looking at the `systemctl status tailscaled` output, I see:

```
% systemctl status tailscaled
● tailscaled.service - Tailscale node agent
     Loaded: loaded (/etc/systemd/system/tailscaled.service; enabled; preset: ignored)
    Drop-In: /nix/store/41pp3nds0bwyv3h3z8dmh0y2rc0aiixc-system-units/tailscaled.service.d
             └─overrides.conf
     Active: active (running) since Thu 2025-05-29 15:47:56 CEST; 18s ago
 Invocation: 8bbe92a80cb9452390e8db0dd6346ff6
       Docs: https://tailscale.com/kb/
   Main PID: 768 (.tailscaled-wra)
     Status: "Stopped; run 'tailscale up' to log in"
```

So, the status message confirms my understanding: tailscale CLI commands are stateful. 

> I think it's more of configuration/documentation issue.

What documentation specifically would you want to see improved? https://tailscale.com/kb/1080/cli#down clearly describes:

> Disconnect from Tailscale. Running this command is the same as choosing to disconnect from or quit a Tailscale client.
> When disconnected, you cannot reach devices over Tailscale. **To reconnect, re-run tailscale up without any flags.**

(Notably, it doesn’t say anything about the node automatically reconnecting.)

> But isn't it weird that the (first) automatic 'tailscale up' on startup does not fix the situation, whereas my (second) manual tailscale up from the web UI does?

Hm, I suspect the tailscale CLI starts up more quickly than the tailscaled daemon, and maybe the CLI does not wait for the daemon to start up? It might be worth filing an issue for this in the tailscale issue tracker. You could work around it by running a custom program that runs `time.Sleep(5*time.Second)` before exec'ing `tailscale up`.

mpl, 2025-05-29:
> I just tested the behavior on NixOS 25.05: After `tailscale down`, the node does not come back after a reboot. Looking at the `systemctl status tailscaled` output, I see:
>
> ```
> % systemctl status [...]

oh yes, sorry, I didn't report back, but I had come to the same conclusion after some more testing.

> What documentation specifically would you want to see improved? https://tailscale.com/kb/1080/cli#down clearly describes:
>
>
> (Notably, it doesn’t say anything about the node automatically [...]

Even after reading that, I had assumed that "up", and "down" are stateless. Kindof like a "start" and "stop" for a service manager are maybe? On one hand they're doing a good job at not bothering the user with describing how the state is kept, but on the other hand, if it was explained there perhaps I would not have been confused. So I guess that's what I'd do (I'd link to another doc page where they explain how state is persisted on tailscale).
As a software writer, I would say that if at least one of my user was confused (in that case myself) about a behaviour, then there might be something to be improved. :-)
Anyway, I did like you do (turned off tailscaled itself when testing), and all went well for me. I'm almost done with my mix of tailscale subnet router + manual gateway with nftables for my project.

> Hm, I suspect the tailscale CLI starts up more quickly than the tailscaled daemon, and maybe the CLI does not wait for the daemon to start up? It might be worth filing an issue for this in the [...]

agreed, that was my prime suspicion too.
Even without the `tailscaled down` business, there has been a few rare times when I set up a gokrazy instance, and tailscale was not working, and manually doing a "run once" of `tailscale up` fixed the issue. So probably the same problem at play.

Maybe we want to add that bit about waiting a few seconds, around the examples on the Gokrazy tailscale page?

closed 2025-05-30 by stapelberg

stapelberg, 2025-06-07:
> On a related note, I'm sure you have thought about what I'm about to mention already: what's your opinion/plan about inter services dependencies? I know we already have things like "waiting for clock", and "waiting for network to be up", but what about service A needing to wait for service B before starting? 

Currently, gokrazy just supervises/restarts services indefinitely, meaning they will ultimately settle and fulfill their dependencies. What would more fine-grained dependency specification provide in addition? 

> I know it wouldn't work for our tailscale VS tailscaled case, because we don't want to wait for tailscaled to be over, we want to wait to it to be "ready", which we don't have any hint about.

I think on systemd-managed systems, tailscaled declares a listening socket and then, when the `tailscale` CLI connects, the system will lazily start `tailscaled` and the initial connection attempt will succeed (despite being a little slow).

socket activation is a pretty elegant concept (I have also used it in the i3 test suite with great success, see https://i3wm.org/docs/testsuite.html#_appendix_c_socket_activation), so maybe one answer would be to implement socket activation.

But, thinking about this again, I wonder if the exit codes of tailscale up are correct: wouldn’t gokrazy restart the tailscale cli until it succeeds? Maybe we’d need to fix / think more about gokrazy’s exit code behavior. 

> But it still makes me think about inter services dependencies, like services manager usually have.

FWIW, the concept of inter-service dependencies, or even dependencies in general, is far from solved in service managers. Just the other day I ran into an issue where I have a daily sync jobs, but specifying `After=network.target` is not sufficient to make it wait for the network to be up *after resuming from suspend-to-RAM*. Instead, I’ll need to explicitly poll IPv4/IPv6 connectivity, or handle errors better (retry the sync N times?). But the service manager’s dependencies don’t really help :-/

> I actually already hit the case for my main project, where I need one of the services to have already accomplished one task before the others can start. So I simply added an net/rpc listener in this one service, and all the others services routinely probe it until the main one replies with an answer indicating it's ready. This was easy and is actually working well enough that I can replicate it if I ever need to do it again, but it made me wonder if there could be something like that at the supervisor level of gokrazy.

The pattern to poll for healthiness/readiness is a good one. I do the same in my backup coordinator, where I use wake-on-LAN to wake up my network storage PC and use polling to detect when it’s up and running and has mounted its disks.