gokrazy/gokrazy #325: TLS error

BrunoTeixeira1996, 2025-07-01:
After updating my config.json to use TLS i have the following error
```
2025/07/01 18:18:16 Including loadable kernel modules from:
2025/07/01 18:18:16   /root/go/pkg/mod/github.com/gokrazy/kernel.rpi@v0.0.0-20250520195459-906287a872d7/lib/modules
2025/07/01 18:18:17 Using certificate /root/.config/gokrazy/hosts/brun0-pi/cert.pem
[done] in 0.06s
ERROR:
  checking target partuuid support: Get "https://gokrazy:***@brun0-pi:1080/update/features": http: server gave HTTP response to HTTPS client
```

config.json
```
{
    "Hostname": "brun0-pi",
    "Update": {
        "HTTPPassword": "____________",
        "HTTPPort": "1080",
        "UseTLS": "self-signed",
        "HTTPSPort": "8443"
    },
```

i've used `--insecure` with `gok update` but i am having this error

stapelberg, 2025-12-17:
Are you still setting `InternalCompatibilityFlags` in `config.json`? If so, please remove that part and try again and post the full output.

BrunoTeixeira1996, 2025-12-17:
> So can you post your current config.json and the reproduction steps please? Including full output.

```
{
    "Hostname": "brun0-pi",
    "Update":
    {
        "HTTPPassword": "REDACTED",
        "HTTPPort": "1080"
    },
    "Environment":
    [
        "GOOS=linux",
        "GOARCH=arm64"
    ],
    "Packages":
    [
        "github.com/gokrazy/fbstatus",
        "github.com/gokrazy/hello",
        "github.com/gokrazy/serial-busybox",
        "github.com/gokrazy/breakglass",
        "github.com/gokrazy/wifi",
        "github.com/gokrazy/mkfs",
        "tailscale.com/cmd/tailscaled",
        "tailscale.com/cmd/tailscale",
        "github.com/BrunoTeixeira1996/waiw/cmd/waiw",
        "github.com/prometheus/node_exporter",
        "github.com/gokrazy/rsync/cmd/gokr-rsync",
        "github.com/TheMightyGit/rssole/cmd/rssole",
        "github.com/gokrazy/stat/cmd/gokr-webstat",
        "github.com/gokrazy/stat/cmd/gokr-stat",
        "github.com/BrunoTeixeira1996/gmah/cmd/gmah",
        "github.com/caddyserver/caddy/v2/cmd/caddy"
    ],
    "PackageConfig":
    {
        "github.com/gokrazy/breakglass":
        {
            "ExtraFilePaths":
            {
                "/etc/breakglass.authorized_keys": "breakglass.authorized_keys"
            },
            "CommandLineFlags":
            [
                "-authorized_keys=/etc/breakglass.authorized_keys"
            ]
        },
        "github.com/gokrazy/gokrazy/cmd/randomd":
        {
            "ExtraFileContents":
            {
                "/etc/machine-id": "d5a1f859e3314622b8a76f12489ce2f4\n"
            }
        },
        "github.com/gokrazy/wifi":
        {
            "ExtraFilePaths":
            {
                "/etc/wifi.json": "wifi.json"
            }
        },
        "github.com/caddyserver/caddy/v2/cmd/caddy":
        {
            "CommandLineFlags":
            [
                "run",
                "--config",
                "/etc/caddy/Caddyfile"
            ],
            "ExtraFilePaths":
            {
                "/etc/caddy/Caddyfile": "Caddyfile"
            }
        },
        "github.com/BrunoTeixeira1996/gmah/cmd/gmah":
        {
            "CommandLineFlags":
            [
                "-email=REDACTED",
                "-password=REDACTED",
                "-dump=/perm/home/gmah/html/",
                "-gokrazy"
            ],
            "WaitForClock": true,
            "ExtraFilePaths":
            {
                "/etc/gmah/html": "/root/gokrazy/brun0-pi/gmah/html",
                "/etc/gmah/serve_template.html": "/root/gokrazy/brun0-pi/gmah/serve_template.html"
            }
        },
        "github.com/BrunoTeixeira1996/waiw/cmd/waiw":
        {
            "CommandLineFlags":
            [
                "-gokrazy",
                "-ip='192.168.30.20'",
                "-user='REDACTED'",
                "-password='REDACTED'",
                "-dbname='REDACTED'"
            ],
            "ExtraFilePaths":
            {
                "/etc/waiw/assets": "/root/personal/waiw/assets",
                "/etc/waiw/templates": "/root/personal/waiw/templates"
            }
        },
        "github.com/TheMightyGit/rssole/cmd/rssole":
        {
            "CommandLineFlags":
            [
                "-gokrazy",
                "-c=/perm/home/rssole/feeds.json",
                "-r=/perm/home/rssole/readcache.json"
            ],
            "WaitForClock": true,
            "ExtraFilePaths":
            {
                "/etc/rssole/feeds.json": "/root/gokrazy/brun0-pi/rssole/feeds.json",
                "/etc/rssole/readcache.json": "/root/gokrazy/brun0-pi/rssole/readcache.json"
            }
        },
        "github.com/gokrazy/rsync/cmd/gokr-rsync":
        {
            "CommandLineFlags":
            [
                "--daemon",
                "--gokr.config=/etc/gokr-rsyncd.toml"
            ],
            "ExtraFilePaths":
            {
                "/etc/gokr-rsyncd.authorized_keys": "gokr-rsyncd.authorized_keys",
                "/etc/gokr-rsyncd.toml": "gokr-rsyncd.toml"
            }
        },
        "github.com/gokrazy/stat/cmd/gokr-webstat":
        {
            "CommandLineFlags":
            [
                "-modules=cpu,disk,sys,net,mem,thermal"
            ]
        },
        "tailscale.com/cmd/tailscale":
        {
            "CommandLineFlags":
            [
                "up",
                "--advertise-routes=192.168.30.0/24"
            ]
        }
    },
    "SerialConsole": "serial0,115200"
```

I can't show the full output right now however I noticed one thing.
I control my gokrazy from 2 servers:
- LXC in proxmox (this is the regular server that controls Gokrazy)
- Personal laptop (this only controls Gokrazy to `gok overwrite` when needed)

with this in mind I noticed that the LXC (where I mostly config everything) contained the following
```
root@gkconfig:~/.config/gokrazy/hosts/brun0-pi# ls
cert.pem  key.pem
```

The laptop did not contained this so I deleted these two files from LXC and it worked again
Maybe `gok` is verifying if these files exist first and thinks TLS is still being used instead of looking at `config.json` ?

stackcoder, 2026-01-07:
While updating using another computer I ran into following error:

First, kinda expected:
> checking target partuuid support: Get "https://gokrazy:***@mygok/update/features": tls: failed to verify certificate: x509: certificate signed by unknown authority (possibly because of "x509: invalid signature: parent certificate cannot sign this kind of certificate" while trying to verify candidate authority certificate "gokrazy")

Second, using `--insecure` flag:
> updating root file system: unexpected HTTP status code: got 400 Bad Request, want 200 (body "expected a PUT request\n")

I was able to update by manually forcing to https using:
```json
"InternalCompatibilityFlags": {
  "Update":"https://gokrazy:***@mygok"
}
```